tlace17/K1-Linear-Rail-Gantry #6: Y-axis linear rails

amr-labib, 2024-10-15:
Hi,

Are there any plans for a version with Y-axis Linear rails?

Thanks,
Labib

ACKET2009, 2024-12-02:
The original creality rods is 8mm and they vibrate during the testing of X axis for input shaper... It will be better to place the X rodas to the Y position... IMHO

tlace17, 2024-12-02:
Linear Rails on Y were trials for over 4 months and due to the flexible nature of the creality frame they cause large racking issues and binding, therefore the work was scrapped.

closed 2024-12-02 by tlace17

ACKET2009, 2024-12-02:
I find this source https://github.com/kemsky/creality-k1-max-xy-rails-mod maybe it will help...

tlace17, 2024-12-02:
> I find this source https://github.com/kemsky/creality-k1-max-xy-rails-mod maybe it will help...

This requires modification of the K1 frame, this is something that is beyond the scope of what I was trying to accomplish. This mod was intended to be accessible to people without physical modification to any part of your printer.