tlace17/K1-Linear-Rail-Gantry #9: STEP files not opening

skitty777, 2025-01-26:
I downloaded all step files but none are working in SolidWorks or Fusion. 
Says files headers are corrupt.

tlace17, 2025-01-26:
Hmm. I'll take a look.

tlace17, 2025-01-27:
I will leave this open for others to comment on if they have the same issue, although I tried 3 different CAD programs with no issues including Fusion and Solidworks

skitty777, 2025-01-27:
Still not working for me. Error messages posted below. Also tried in Fusion with same result. 

[image]
[image]

skitty777, 2025-01-27:
All recently downloaded STL files are failing too! 

[image]

closed 2025-01-27 by skitty777

iEBPO, 2025-01-29:
The screenshot shows that the files were downloaded with an error. I tried in Fusion files open without problems.

[image]

tlace17, 2025-01-29:
He figured it out, just hade to clear his browser data and it seemed to fix it for him.